turtlebot/turtlebot #186: Errors in minimal.launch using Linux Mint.

spragunr, 2015-01-07:
I'm trying out the turtlebot packages currently in shadow-fixed, and it looks like the app_manager(?) is hard-coded to expect Ubuntu.  I've used Turtlebot with Linux Mint since Fuerte, and I've never had any trouble before.  The relevant packages are the same.   

If I force packages.linux_distribution() to return ('Ubuntu', '14.04', 'Trusty'), the problem goes away, and I can launch and telop the Turtlebot.  The error messages are reproduced below:  

```
... logging to /home/spragunr/.ros/log/4cb70eaa-968f-11e4-855a-ecf4bb1d9aca/roslaunch-tb-test-s15-6820.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.
started roslaunch server http://tb-test-s15:49790/

SUMMARY
========

PARAMETERS
 * /app_manager/auto_rapp_installation: False
 * /app_manager/auto_start_rapp: 
 * /app_manager/capability_server_name: capability_server
 * /app_manager/local_remote_controllers_only: False
 * /app_manager/rapp_package_blacklist: []
 * /app_manager/rapp_package_whitelist: ['rocon_apps', 't...
 * /app_manager/remote_controller_blacklist: []
 * /app_manager/remote_controller_whitelist: []
 * /app_manager/robot_icon: turtlebot_bringup...
 * /app_manager/robot_name: turtlebot
 * /app_manager/robot_type: turtlebot
 * /app_manager/screen: True
 * /app_manager/simulation: False
 * /bumper2pointcloud/pointcloud_radius: 0.24
 * /capability_server/blacklist: ['std_capabilitie...
 * /capability_server/defaults/kobuki_capabilities/KobukiBringup: kobuki_capabiliti...
 * /capability_server/defaults/kobuki_capabilities/KobukiBumper: kobuki_capabiliti...
 * /capability_server/defaults/kobuki_capabilities/KobukiCliffDetection: kobuki_capabiliti...
 * /capability_server/defaults/kobuki_capabilities/KobukiLED1: kobuki_capabiliti...
 * /capability_server/defaults/kobuki_capabilities/KobukiLED2: kobuki_capabiliti...
 * /capability_server/defaults/kobuki_capabilities/KobukiLED: kobuki_capabiliti...
 * /capability_server/defaults/kobuki_capabilities/KobukiWheelDropDetection: kobuki_capabiliti...
 * /capability_server/defaults/std_capabilities/Diagnostics: turtlebot_capabil...
 * /capability_server/defaults/std_capabilities/DifferentialMobileBase: kobuki_capabiliti...
 * /capability_server/defaults/std_capabilities/LaserSensor: turtlebot_capabil...
 * /capability_server/defaults/std_capabilities/RGBDSensor: turtlebot_capabil...
 * /capability_server/defaults/std_capabilities/RobotStatePublisher: turtlebot_capabil...
 * /capability_server/defaults/turtlebot_capabilities/TurtleBotBringup: turtlebot_capabil...
 * /capability_server/nodelet_manager_name: capability_server...
 * /capability_server/package_whitelist: ['kobuki_capabili...
 * /cmd_vel_mux/yaml_cfg_file: /opt/ros/indigo/s...
 * /description: Kick-ass ROS turtle
 * /diagnostic_aggregator/analyzers/input_ports/contains: ['Digital Input',...
 * /diagnostic_aggregator/analyzers/input_ports/path: Input Ports
 * /diagnostic_aggregator/analyzers/input_ports/remove_prefix: mobile_base_nodel...
 * /diagnostic_aggregator/analyzers/input_ports/timeout: 5.0
 * /diagnostic_aggregator/analyzers/input_ports/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/kobuki/contains: ['Watchdog', 'Mot...
 * /diagnostic_aggregator/analyzers/kobuki/path: Kobuki
 * /diagnostic_aggregator/analyzers/kobuki/remove_prefix: mobile_base_nodel...
 * /diagnostic_aggregator/analyzers/kobuki/timeout: 5.0
 * /diagnostic_aggregator/analyzers/kobuki/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/power/contains: ['Battery', 'Lapt...
 * /diagnostic_aggregator/analyzers/power/path: Power System
 * /diagnostic_aggregator/analyzers/power/remove_prefix: mobile_base_nodel...
 * /diagnostic_aggregator/analyzers/power/timeout: 5.0
 * /diagnostic_aggregator/analyzers/power/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/sensors/contains: ['Cliff Sensor', ...
 * /diagnostic_aggregator/analyzers/sensors/path: Sensors
 * /diagnostic_aggregator/analyzers/sensors/remove_prefix: mobile_base_nodel...
 * /diagnostic_aggregator/analyzers/sensors/timeout: 5.0
 * /diagnostic_aggregator/analyzers/sensors/type: diagnostic_aggreg...
 * /diagnostic_aggregator/base_path: 
 * /diagnostic_aggregator/pub_rate: 1.0
 * /icon: turtlebot_bringup...
[... 80 lines cut ...]
process[bumper2pointcloud-7]: started with pid [7041]
process[turtlebot_laptop_battery-8]: started with pid [7123]
process[capability_server-9]: started with pid [7132]
process[app_manager-10]: started with pid [7135]
process[master-11]: started with pid [7140]
process[interactions-12]: started with pid [7145]
process[zeroconf/zeroconf-13]: started with pid [7207]
[zeroconf/zeroconf-13] killing on exit
[interactions-12] killing on exit
[master-11] killing on exit
[capability_server-9] killing on exit
[turtlebot_laptop_battery-8] killing on exit
[bumper2pointcloud-7] killing on exit
[cmd_vel_mux-6] killing on exit
[mobile_base-5] killing on exit
[mobile_base_nodelet_manager-4] killing on exit
[diagnostic_aggregator-3] killing on exit
[robot_state_publisher-2] killing on exit
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

jihoonl, 2015-01-08:
`mint` was not listed in rocon_uri rule. Now it should have been fixed by

https://github.com/robotics-in-concert/rocon_tools/commit/bdddc84b58e550b0bf4536c11bc5e807ae743fed.

Please reopen if it still happens.

jihoonl, 2015-01-08:
My bad. The logic in [rocon_app_utilities/DependencyChecker](https://github.com/robotics-in-concert/rocon_app_platform/blob/indigo/rocon_app_utilities/src/rocon_app_utilities/dependencies.py#L48-L57) should be fixed.

jihoonl, 2015-01-08:
@spragunr Can you tell what you get from rospkg.os_detect.OsDetect().get_codename()?

jihoonl, 2015-01-08:
The problem is that [rocon_app_utilities.DependencyChecker](https://github.com/robotics-in-concert/rocon_app_platform/blob/indigo/rocon_app_utilities/src/rocon_app_utilities/dependencies.py#L48-L60) uses a custom way to detect operating system. I think it is supposed to use rospkg.os_detect module. 

Ref : https://github.com/ros-infrastructure/rosdep/blob/master/src/rosdep2/lookup.py#L86

jihoonl, 2015-01-08:
@spragunr, I have patched dependency check with https://github.com/robotics-in-concert/rocon_app_platform/issues/269. 

Could you confirm that mint works with source version? Then I will release the patch.

spragunr, 2015-01-08:
```
In [8]: rospkg.os_detect.OsDetect().get_codename()
Out[8]: 'rebecca'

In [9]: rospkg.os_detect.OsDetect().get_version()
Out[9]: '17.1'

In [10]: rospkg.os_detect.OsDetect().get_name()
Out[10]: 'mint'
```
 
If you are going to store these in a config file somewhere, you should also include:

```
In [8]: rospkg.os_detect.OsDetect().get_codename()
Out[8]: 'qiana'

In [9]: rospkg.os_detect.OsDetect().get_version()
Out[9]: '17'

In [10]: rospkg.os_detect.OsDetect().get_name()
Out[10]: 'mint'
```
 Both qiana and rebecca are derived from Ubuntu Trusty.

spragunr, 2015-01-08:
It doesn't look like the changes to  robotics-in-concert/rocon_app_platform solve the problem.  I get the same error message with the source version: 

```
Traceback (most recent call last):
  File "/home/spragunr/catkin_ws/src/rocon_app_manager/scripts/rapp_manager.py", line 19, in <module>
    manager = rocon_app_manager.RappManager()
  File "/home/spragunr/catkin_ws/src/rocon_app_manager/src/rocon_app_manager/rapp_manager.py", line 68, in __init__
    self._dependency_checker = rocon_app_utilities.DependencyChecker(self._indexer)
  File "/home/spragunr/catkin_ws/src/rocon_app_platform/rocon_app_utilities/src/rocon_app_utilities/dependencies.py", line 61, in __init__
    self.installer_keys = self.installer_context.get_os_installer_keys(self.os_name)
  File "/usr/lib/python2.7/dist-packages/rosdep2/installers.py", line 199, in get_os_installer_keys
    raise KeyError(os_key)
KeyError: 'mint'
[app_manager-10] process has died [pid 1014, exit code 1, cmd /home/spragunr/catkin_ws/src/rocon_app_manager/scripts/rapp_manager.py app_manager/gateway_info:=gateway/gateway_info app_manager/remote_gateway_info:=gateway/remote_gateway_info app_manager/force_update:=gateway/force_update app_manager/flip:=gateway/flip __name:=app_manager __log:=/home/spragunr/.ros/log/867f6952-9757-11e4-9083-ecf4bb1d9aca/app_manager-10.log].
log file: /home/spragunr/.ros/log/867f6952-9757-11e4-9083-ecf4bb1d9aca/app_manager-10*.log
```

What does the rocon app manager actually do?  The Turtlebot seems to function correctly even though this node dies.  As a workaround, can I just ignore it?   If that is the case, then I think that content should be removed from minimal.launch.  As much as possible, minimal.launch should include only the minimum set of nodes necessary to interact with the robot.

stonier, 2015-01-09:
Does rosdep actually work on Mint? That seems to be the core of the problem here. If that's true, we can certainly add an exception for KeyErrors here to disable installation functionality in the rocon app manager.

| What does the rocon app manager actually do? 

Previously you had to *magically know* what launchers are required to launch what activity, which can be confusing as there are often many launchers lying around and no central 'list' for them (basically you have to rely on trawling the wiki). The rocon app manager instead loads the correct list on startup and manages the launching them for you. It gets even easier once you connect an android to it. We never used to be able to get our sales/marketing guys able to demos, but we can like this. Even internally as devs we often find it easier working like this - especially if we just want to teleop the robot.

We support the roslaunch way still as that's an important part of learning how to be a ros dev. But the rocon app manager also lets us support a different mode of access as described above. If you don't make use of it, it will just idle so it doesn't significantly get in the way.

jihoonl, 2015-01-09:
I made a patch to disable rapp installer in `rocon_app_manager`. so you can use turtlebot without any error in app_manager.

Could you test with [installer_disable branch](https://github.com/robotics-in-concert/rocon_app_platform/tree/installer_disable) to see if it works in mint?

It would only give you a warning that auto rapp installation is disabled.

spragunr, 2015-01-09:
@jihoonl  This branch does prevent the crash.  There are still a large number of warnings of the form: 
```
[WARN] [WallTime: 1420808870.826084] Rapp Manager : 'turtlebot_rapps/xbox360_teleop' is invalid [operating_system specification is invalid or not officially registered (see https://github.com/robotics-in-concert/rocon_tools/blob/indigo/rocon_uri/src/rocon_uri/rules/rules.yaml)[rebecca]]
```
It's unfortunate, but a good-enough workaround for now.  If the underlying issue is a lack of mint support for rosdep, I suppose that's the correct place to address this.  Thanks for working on it.

spragunr, 2015-01-09:
@stonier Currently rosdep only works on Mint if you explicitly override the os detection with the --os flag: 
```
rosdep --os ubuntu:trusty ...
```

Thanks for the app-manager clarification. I definitely see the value in what you are describing.

> We support the roslaunch way still as that's an important part of learning how to be a ros dev. But the rocon app manager also lets us support a different mode of access as described above. If you don't make use of it, it will just idle so it doesn't significantly get in the way.

From an educational point of view, it's helpful to be able to reveal levels of abstraction incrementally.   Even if the app manager doesn't get in the way from a functional point of view, there is cognitive overhead in the form of additional notifications in the terminal and more complex launch files.  Unless there is some technical reason that makes it difficult, my preference would be for minimal.launch to be as true to the name as possible.   Could there be another file (something like rapp.launch) that includes minimal.launch, then brings up the app manager?

stonier, 2015-01-10:
I agree, what I would like is tutorials start by showing how you build up the layers for a task, i.e. shutdown all currently running launchers and start with nothing running, then base.launch, then robot.launch, to the point where you actually have the robot performing a task. At this point its starting to look complicated, so switch the tutorials to starting the rapp manager and simply *start task*.

But we have to deliver a working system that has a runtime that works out of the box. Having to start half a dozen extra launchers to get to the mechanisms which save you having to start half a dozen extra launchers would be defeating the point.

Ironically...we have discussed minimal.launch as just launching the rapp manager :neckbeard:

stonier, 2015-01-10:
I added `rebecca` to the [rocon uri rules](https://github.com/robotics-in-concert/rocon_tools/blob/indigo/rocon_uri/src/rocon_uri/rules/rules.yaml). In future you can edit/pull request that for mint specifics.

sosentos, 2015-01-14:
We are seeing a similar issue with linaro.  We are trying to get an ARM board working with the Turtlebot and are currently using the current deb packages.

1)  We installed the turtlebot packages from debs.  We then tried roslaunch turtlebot_bringup minimal.launch and see the following error:
rospkg.os_detect.OsNotDetected: Could not detect OS, tried ['windows', 'ubuntu', 'rhel', 'qnx', 'osx', 'opensuse', 'mint', 'gentoo', 'freebsd', 'fedora', 'debian', 'cygwin', 'arch']

2)  If we export ROS_OS_OVERRIDE=ubuntu:trusty or modify the os_detect.py file to include linaro we see the following errors:

self.installer_keys = self.installer_context.get_os_installer_keys(self.os_name) 

File "/usr/lib/pymodules/python2.7/rosdep2/installers.py", line 206, in get_os_installer_keys 

raise KeyError(os_key)
KeyError: 'linaro

stonier, 2015-01-14:
1. Can you paste the exact output you get from 1)? Is it an exception that is getting raised (in which case the backtrace is good) or just information going to stdout?

2. The KeyError should not be caught and gracefully handled when Jihoon gets a next release out.

sosentos, 2015-01-14:
I can do it tomorrow when I have the machine in front of me.  

If I remember right an exception came from line 488 of os_detect.py in rospkg.  I believe it was called from line 518 of the same file.

jihoonl, 2015-01-14:
The new version 0.7.6 is now available in shadow-fixed. @sosentos could you check it still gives you an error?

jihoonl, 2015-01-14:
https://github.com/robotics-in-concert/rocon_app_platform/pull/270 this PR should prevent KeyError and disable auto rapp installer.

stonier, 2015-01-14:
@jihoonl looks like we need to catch `rospkg.os_detect.OsNotDetected`

@sosentos if you can get us a backtrace, we'll know exactly where to place the catcher.

sosentos, 2015-01-14:
Here is the full traceback
```
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/rocon_app_manager/rapp_manager.py", line 19, in <module>
    manager = rocon_app_manager.RappManager()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rocon_app_manager/rapp_manager.py", line 61, in __init__
    (self._rocon_uri, self._icon) = self._set_platform_info()
  File "/opt/ros/indigo/lib/python2.7/dist-packages/rocon_app_manager/rapp_manager.py", line 92, in _set_platform_info
    os_codename = rospkg.os_detect.OsDetect().get_codename()
  File "/usr/lib/pymodules/python2.7/rospkg/os_detect.py", line 499, in get_codename
    self.detect_os()
  File "/usr/lib/pymodules/python2.7/rospkg/os_detect.py", line 459, in detect_os
    raise OsNotDetected("Could not detect OS, tried %s"%attempted)
rospkg.os_detect.OsNotDetected: Could not detect OS, tried ['windows', 'ubuntu', 'rhel', 'qnx', 'osx', 'opensuse', 'mint', 'gentoo', 'freebsd', 'fedora', 'debian', 'cygwin', 'arch']
[ERROR] [WallTime: 1421260385.087815] Interactions : timed out trying to find the rapp manager start_rapp, stop_rapp services and status topic [unable to communicate with the master [no provider]]```

jihoonl, 2015-01-15:
It looks like we should update [a list of OS in rospkg.os_detect](https://github.com/ros-infrastructure/rospkg/blob/master/src/rospkg/os_detect.py#L531-L561) to detect `linaro` as supported os.

Can you provide what you get from the following command? 

```
$ python
Python 2.7.6 (default, Mar 22 2014, 22:59:56) 
[GCC 4.8.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import platform
>>> platform.linux_distribution()
```

It should give us what to add in the list.

sosentos, 2015-01-15:
('Linaro', '14.04', 'trusty')

jihoonl, 2015-01-15:
I have created PR to update OS list in rospkg. Once it goes through linaro should be fine.

jihoonl, 2015-01-15:
Hi @spragunr,

you may want to chase up https://github.com/ros-infrastructure/rospkg/issues/25 if you need a proper mint support in ROS.

jihoonl, 2015-03-24:
This should have been resolved. Please reopen if necessary.

closed 2015-03-24 by jihoonl