turtlebot/turtlebot #197: Remaining turtleobt tutorial works

jihoonl, 2015-03-16:
[image]

jihoonl, 2015-03-16:
* [ ] Fix chirp meow -> moo
* [ ] fix `no preffered rapp` warning
* [ ] create rapp tutorial
* [ ] create interaction tutorial
* [ ] flag a multi robot tutorial
* [ ] stage/gazebo tutorial link
* [ ] check 3d visualization for depth image/registration - explain `either` works.

stonier, 2015-03-17:
Return to links should be:
```
[[Robots/TurtleBot|TurtleBot main page|&distro=indigo]]
```
not
```
[[Robots/TurtleBot|TurtleBot main page]]
```

stonier, 2015-03-20:
@jihoonl can you update with current status?

jihoonl, 2015-03-20:
Done jobs are checked.

jihoonl, 2015-03-20:
Flag a multi robot tutorial?

stonier, 2015-03-20:
Empty page, with a TODO and we come back after the rocon documentation.

jihoonl, 2015-04-15:
Done.

closed 2015-04-15 by jihoonl