turtlebot/turtlebot #206: turtlebot_joy.cpp is generic enough

130s, 2015-03-30:
(With a risk of hitting FAQ) For example I can use the node in my own package without specifying anything specific to Turtlebot ([see this](https://github.com/tork-a/spur/pull/15/files)). 
How about moving it to somewhere common, eg. https://github.com/ros-drivers/joystick_drivers?

mikeferguson, 2015-03-30:
If it were going to be merged with something, it should be merged with https://github.com/ros-teleop/teleop_twist_joy

mikeferguson, 2015-03-30:
Also, I think teleop_twist_joy is mainly missing the mux support (as far as being a drop in replacement)

mmwise, 2015-03-30:
I think I am torn by this largely because TurtleBot is meant to be instructional and stable. 

In terms of instructional all the code is in one place for people to look at and understand what is going on and has several options configured specifically for te TurtleBot. However it may model ROS inproperly by making yet another teleop package... 

In terms of stability the TurtleBot tutorials all depend on these launch files etc and would then need to be maintained/changed to reflect the locations of the new launch files. Along with updating other higher level launch files.

tfoote, 2015-03-30:
I think it's fine to keep them separate for now. Migrating it will be a large effort and there's a lot of references etc to the current state which is not broken and does not have a pressing need to change in itself.

If you want specific functionality provided by turtlebot_joy I'd say go ahead and use it, or better submit a pull-request to add that capability to teleop_twist_joy. I agree that creating yet another teleop package does not make sense.

If teleop_twist_joy reaches the superset of turtlebot_joy we can choose to switch the implementation, or update the launch files in turtlebot packages.

130s, 2015-04-28:
All comments make sense to me. I switched my application to https://github.com/ros-teleop/teleop_twist_joy as well. Thanks!

closed 2015-04-28 by 130s