turtlebot/turtlebot #211: Got this error while using fake_turtlebot.launch in simple_gui.html

Shrawan99, 2015-06-19:
Invalid <arg> tag: rbx1_description
ROS path [0]=/opt/ros/hydro/share/ros
ROS path [1]=/home/turtlebot/catkin_ws/src
ROS path [2]=/opt/ros/hydro/share
ROS path [3]=/opt/ros/hydro/stacks. 

Arg xml is 
<arg default="$(find xacro)/xacro.py '$(find rbx1_description)/urdf/turtlebot.urdf.xacro'" name="urdf_file"/>

**fake_turtlebot.launch**

<launch>

  <param name="/use_sim_time" value="false" />

  
  <arg name="battery_runtime" default="60" />
  
  
  <arg name="map" default="blank_map.yaml" />

  
  <arg name="urdf_file" default="$(find xacro)/xacro.py '$(find rbx1_description)/urdf/turtlebot.urdf.xacro'" />
   
  <param name="robot_description" command="$(arg urdf_file)" />
    
  <node name="arbotix" pkg="arbotix_python" type="arbotix_driver" clear_params="true" output="screen">
     <rosparam file="$(find rbx1_bringup)/config/fake_turtlebot_arbotix.yaml" command="load" />
     <param name="sim" value="true"/>
  </node>
  
  <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher">
      <param name="publish_frequency" type="double" value="20.0" />
  </node>
  
  
  <node pkg="tf" type="static_transform_publisher" name="odom_left_wheel_broadcaster" args="0 0 0 0 0 0 /base_link /left_wheel_link 100" />
  <node pkg="tf" type="static_transform_publisher" name="odom_right_wheel_broadcaster" args="0 0 0 0 0 0 /base_link /right_wheel_link 100" />
  
  
  <node name="map_server" pkg="map_server" type="map_server" args="$(find rbx2_nav)/maps/$(arg map)" />
  
  
  <node pkg="move_base" type="move_base" respawn="false" name="move_base" clear_params="true" output="screen">
     <rosparam file="$(find rbx2_nav)/config/fake/costmap_common_params.yaml" command="load" ns="global_costmap" />
     <rosparam file="$(find rbx2_nav)/config/fake/costmap_common_params.yaml" command="load" ns="local_costmap" />
     <rosparam file="$(find rbx2_nav)/config/fake/local_costmap_params.yaml" command="load" />
     <rosparam file="$(find rbx2_nav)/config/fake/global_costmap_params.yaml" command="load" />
     <rosparam file="$(find rbx2_nav)/config/fake/base_local_planner_params.yaml" command="load" />
     <rosparam file="$(find rbx2_nav)/config/nav_test_params.yaml" command="load" />

  </node>
  
  <include file="$(find rbx2_utils)/launch/battery_simulator.launch">
     <arg name="battery_runtime" value="$(arg battery_runtime)" />
  </include>
  
  
  <node pkg="fake_localization" type="fake_localization" name="fake_localization" output="screen" />
      
  
  <node pkg="tf" type="static_transform_publisher" name="odom_map_broadcaster" args="0 0 0 0 0 0 /map /odom 100" />
  
</launch>

trainman419, 2015-06-19:
Please reformat your question to use the preformatted text option; as it is all of the important data in your issue isn't visible.

Shrawan99, 2015-06-19:
Sorry sir i am new  on posting question in Github.. There is some problem in *rbx1_description* ..

trainman419, 2015-06-19:
Please see the github formatting guide: https://help.github.com/articles/github-flavored-markdown/

trainman419, 2015-06-19:
This issue tracker is for bugs in the turtlebot code; it looks like you're having trouble with a launch file that you either wrote yourself, or which is from a different package (perhaps the ROS by Example book?)

If this isn't a bug with one of the nodes or launch files in the turtlebot stack, you should follow the [ROS Support Guidelines](http://wiki.ros.org/Support) and ask a question on [ROS Answers](http://answers.ros.org/)