turtlebot/turtlebot #212: Are there any plans for a Jade release?

koenlek, 2015-07-28:
Hi all,

I use the turtlebot packages for navigation simulation, and I wondered if and when I could expect a Jade version of it. I would like to be able to make use of the improvements that Gazebo 5 offers, which is delivered with Jade...

Best,
Koen

nckswt, 2015-09-08:
I'm really interested in this as well. I'd tried a source install to get the Turtlebot stack working in Jade, and after a full day, I couldn't get it working.

jihoonl, 2015-09-09:
Sorry guys. Due to the some difficulties, we have decided to skip jade debs and support only indigo. However, I will try to help you to get jade source version working. What is causing the problem in jade?

spragunr, 2015-09-10:
This is a real shame.  Turtlebot is currently the best option for ROS in education.  Skipping an entire ROS release is a big blow.  Have you contacted OSRF to see if they would be willing to help out?  I would think they would have an interest in keeping Turtlebot viable.   Can you elaborate a bit on the difficulties that are preventing a Jade debs?

bstag, 2015-09-10:
The indigo is fully ready to go announcement for turtlebot only came out
this week.

mikeferguson, 2015-09-10:
I think you'll find that with the short life cycle of Jade (2 years) versus Indigo (5 years), most hardware vendors will stick with Indigo. Both Indigo and Jade support Ubuntu Trusty which is an LTS (long term support, 5 years), whereas all the other Ubuntu versions that Jade adds support for, over Indigo, are only 9 months of support, which is really too short for rolling out onto hardware.

A simulation-only release in Jade, primarily for newer versions of the simulator, might be a more viable option -- that is what we are doing at Fetch Robotics.

tfoote, 2015-09-10:
@spragunr It's not so much that there are specific difficulties, it's that QAing and testing and releasing everything is a lot of work. Especially with tutorials and teaching material there is a lot of content to update for each release. We recommend using the more stable LTS release for these types of things. That's exactly why the LTS exists to allow parts of the community to choose to iterate less often. 

As @jihoonl mentioned if member of the community are interested in making it work we're happy to support them and if it gets to a point that the packages are releaseable we can do that as well. However we're not planning to make a focused effort to support turtlebot on Jade at the moment.

closed 2015-09-10 by tfoote

spragunr, 2015-09-10:
@tfoote I understand, and I appreciate the hard work that goes into a release.  This particular case is a bit disappointing because I rely quite a bit on gazebo in my courses.  The version of gazebo in Indigo is painfully worse than the version in Jade.

I will say that Indigo is the best release yet for working with Turtlebot.  All of the pieces are working together smoothly for the first time in a long time.

jihoonl, 2015-09-11:
@spragunr Sorry, it seems like my wording confused you. Like @tfoote mentioned above and @stonier mentioned in turtlebot sig, the difficulties are not technical. We just don't have human resource to maintain all materials(e.g wiki, tutorials) around debs for short term releases.

However, I understands that your situation that the big changes in jade gazebo makes you to use jade. And jade debs would be great help for users. So like @tfoote said, if any community members are interested in getting jade turtlebot working, we will be happy to support them.