turtlebot/turtlebot #217: a bug: continuously self rotating

somebodyus, 2016-04-28:
I installed the latest version of turtlebot.
I start the minimal.launch, and opened rviz. I set the fixed frame to `odom`, then I found the whole TF tree is rotating continuously. I haven't given any input for the turtlebot.

tfoote, 2016-04-29:
Please provide more information to allow us to reproduce. Helpful information would include how fast it's spinning, what direction, possibly even a screencast. Logged data would also be helpful . What environment is your turtlebot in? Is it repeatable if so how. What hardware are you running on?

somebodyus, 2016-04-30:
Thank you so much, but the problem is very strange.
The bug is automatically fixed after I installed some updates, specially the Nvidia driver.

tfoote, 2016-04-30:
That is odd. I'm going to close this then. Glad to hear it's fixed.

closed 2016-04-30 by tfoote

stonier, 2016-05-01:
If it's rotating around the yaw axis, it could be the gyro. I can't remember if it occurs for that gyro, but some gyros use the first few seconds to initialise their filters. If you are carrying the robot (which you can do with the kobuki) and subsequently moving it in unexpected directions, it will badly initialise the filter. Resulting in a non-zero rate when you finally put the robot at rest on the ground and a tf tree that is rotating.

I know we've put checks for this on our later robots, but kobuki might not have this. Check if you happen to have the same problem again.

somebodyus, 2016-05-01:
Yes. It is rotating around yaw axis.
The behavior is very similar as the IMU rotating. It once happened a few month ago. I thought it was a driver bug, so I compiled from the source and solved the problem. 
How can I fixed the problem if it happen again?

stonier, 2016-05-01:
Just make sure you turn the robot on when it is on the ground and at rest.  That is all.

stonier, 2016-05-01:
Oh, I just remembered. A very long time ago there was a gyro drift bug in the firmware. Check that you have the latest firmware.

* [Firmware Changelog](https://github.com/yujinrobot/kobuki_core/blob/hydro-devel/kobuki_driver/firmware_changelog.md)
* [Firmware Blobs](http://files.yujinrobot.com/kobuki/firmware/)
* [Upgrade Instructions](http://kobuki.yujinrobot.com/wiki/upgrading-firmware-linux/)