turtlebot/turtlebot #219: Will turtlebot be released & supported on Kinetic?

mkhansen-intel, 2016-05-23:
Our team is using the Turtlebot for testing and development. We know that there wasn't support for the turtlebot in 'Jade' release, but we were expecting that 'Kinetic' would include it. However, looking at the beta release and the build farm, it doesn't appear to be added for Kinetic. Is that correct? Are you planning on adding support for Kinetic?

stonier, 2016-05-23:
It's in the pipeline. You'll notice we've got alot of the underlying packages in already.

mkhansen-intel, 2016-05-24:
Thanks for the reply, do you have an ETA?

tfoote, 2016-05-24:
Higher level packages typically lag while waiting for dependencies to be released. We're pretty close on the turtlebot however. 

Comparing to Indigo, the following repositories are blocking.  I think most of these are within the core TurtleBot maintainers list so I hope to get it out in a matter of weeks if not sooner. 

```
	turtlebot:
		depthimage_to_laserscan
		kobuki
		linux_peripheral_interfaces
		openni2_launch
		rocon_app_platform
		rocon_multimaster
		smart_battery_msgs #<-- dependency removed
		yujin_ocs
	turtlebot_apps:
		depthimage_to_laserscan
		kobuki
		linux_peripheral_interfaces
		openni2_launch
		robot_pose_publisher
		rocon_app_platform
		rocon_multimaster
		smart_battery_msgs  #<-- dependency removed
		turtlebot
		turtlebot_msgs
		warehouse_ros
		world_canvas
		world_canvas_libs
		world_canvas_msgs
		yujin_ocs
```

With most of the packages already released I've been running demos with the upper level packages from source. There are instructions here: http://wiki.ros.org/turtlebot/db410c to reproduce most of the demo we ran this weekend at Maker Faire at the core it's an example of how to build from source ontop of a mostly binary installation. If you're not on the db410c then you can skip a bunch of the workarounds in those instructions.

tfoote, 2016-05-24:
@stonier @jihoonl I pushed on a bunch of the blocking repos. we're down to just a few repos left blocking turtlebot release. Could you take a look at what's releaseable and see if we can get turtlebot out.

stonier, 2016-05-25:
Aye, I aim to keep driving it bit by bit when I can free up a few minutes.

tfoote, 2016-06-15:
We're getting pretty close: 

These packages are ready to release: 

- 	robot_pose_publisher
- 	rocon_multimaster
- 	warehouse_ros
- 	yujin_ocs

After rocon_multimaster, rocon_app_platform will be releasable, and after that kobuki. And that will get us to releasing turtlebot. 

The smart_battery_msgs has been collapsed into sensor_msgs so it goes away. 

I plan to split the warehouse/world_canvas demos out of the default turtlebot_apps repo, which means the only additional dependency in turtlebot_apps is the robot_pose_publisher which is ready to go. I'll ping that.

@stonier I'd love to get this out this week. If you won't have time, could I get access to the release repos to be able to run bloom. I'm guessing most of these repos can release the same version into kinetic as indigo at least for now. 

My dependency walking raw info. 
```
$ ./scripts/check_blocking_repos.py --rosdistro kinetic --comparison indigo --repositories turtlebot turtlebot_apps yujin_ocs kobuki rocon_multimaster rocon_app_platform robot_pose_publisher world_canvas warehouse_ros
The following repos cannot be released because of unreleased dependencies:
	kobuki:
		rocon_app_platform
		rocon_multimaster
		yujin_ocs
	rocon_app_platform:
		rocon_multimaster
	turtlebot:
		kobuki
		rocon_app_platform
		rocon_multimaster
		smart_battery_msgs
		yujin_ocs
	turtlebot_apps:
		kobuki
		robot_pose_publisher
		rocon_app_platform
		rocon_multimaster
		smart_battery_msgs
		turtlebot
		warehouse_ros
		world_canvas
		world_canvas_libs
		world_canvas_msgs
		yujin_ocs
	world_canvas:
		warehouse_ros
		world_canvas_libs
		world_canvas_msgs
The following repos can be released, and are blocking other repos:
	robot_pose_publisher
	rocon_multimaster
	warehouse_ros
	yujin_ocs
```

stonier, 2016-06-20:
@tfoote sorry, got nailed at work the last week/weekend. 

Managed to get `yujin_ocs` out. Stuff in `rocon_multimaster` and `rocon_app_platform` have simplified so the old launchers probably won't work out of the box. I'll have to go back and revisit them and the tutorials to make sure they are still ok. I don't think we'll do any support for multi-robot or android out of the box though - just don't have the cycles.

At any rate, blooming rocon_multimaster now. Hopefully that will get through and I'll be able to bloom rocon_app_platform/kobuki through in the morning.

tfoote, 2016-06-20:
Thanks @stonier 

And I see @130s got a release of warehouse_ros out too. Thanks.

@rctoris could you release robot_pose_publisher?

tfoote, 2016-06-20:
@corot could you make releases of the world_canvas stacks into Kinetic? 

It looks like those are the remaing blockers outside the ones mentioned above. 

```
$ ./scripts/check_blocking_repos.py --rosdistro kinetic --comparison indigo --repositories turtlebot turtlebot_apps rocon_app_platform robot_pose_publisher world_canvas kobuki world_canvas_libs world_canvas_msgs
The following repos cannot be released because of unreleased dependencies:
	kobuki:
		rocon_app_platform
	turtlebot:
		kobuki
		rocon_app_platform
		smart_battery_msgs
	turtlebot_apps:
		kobuki
		robot_pose_publisher
		rocon_app_platform
		smart_battery_msgs
		turtlebot
		world_canvas
		world_canvas_libs
		world_canvas_msgs
	world_canvas:
		world_canvas_libs
		world_canvas_msgs
	world_canvas_libs:
		world_canvas_msgs
The following repos can be released, and are blocking other repos:
	robot_pose_publisher
	rocon_app_platform
	world_canvas_msgs
```

stonier, 2016-06-21:
Hopefully have the rocon_xxx stacks through now. Trying to bloom kobuki now.

Just to give you an idea of eta - I'm off on holiday without a computer until Friday. I will play around on Friday to make sure everything through to kobuki is up and building (will do world canvas if @corot doesn't chime in).

Next week I will be able to spend some small time playing with turtlebot to get it working.

rctoris, 2016-06-22:
@tfoote all set https://github.com/ros/rosdistro/pull/11939

tfoote, 2016-06-22:
Thanks @stonier @rctoris! turtlebot itself is unblocked. I'll do some testing and get an initial release out.

@corot will you have a chance to release world_canvas_* soon? It's the only remaining blocker for turtlebot_apps besides turtlebot. (`smart_battery_msgs` has been replaced between indigo and kinetic.)
```
$ ./scripts/check_blocking_repos.py --rosdistro kinetic --comparison indigo --repositories turtlebot turtlebot_apps rocon_app_platform robot_pose_publisher world_canvas kobuki world_canvas_libs world_canvas_msgs
Ignoring inputs which have already been released:
	kobuki
	robot_pose_publisher
	rocon_app_platform
The following repos cannot be released because of unreleased dependencies:
	turtlebot:
		smart_battery_msgs
	turtlebot_apps:
		smart_battery_msgs
		turtlebot
		world_canvas
		world_canvas_libs
		world_canvas_msgs
	world_canvas:
		world_canvas_libs
		world_canvas_msgs
	world_canvas_libs:
		world_canvas_msgs
The following repos can be released, and are blocking other repos:
	world_canvas_msgs
```

ghindman, 2016-06-29:
Any update on expected timeline for release into Kinetic?

tfoote, 2016-06-30:
The basic capabilities up through `turtlebot_apps` are out as of this morning in the testing repo. http://repositories.ros.org/status_page/ros_kinetic_default.html?q=turtlebot

There's one more part turtlebot_viz before I'd like to close this but I think we're pretty much there.

corot, 2016-06-30:
Hi, sorry for not answering before. Yes, I'll try to release world_canvas_* soon...  but I hope we call soon to the same thing! :speak_no_evil. Beginning of next week, most probably.

tfoote, 2016-06-30:
@corot @stonier helped you out by just rereleasing the previous versions I believe.

tfoote, 2016-07-01:
To close this out I found the following dependencies: 

```
$ ./scripts/check_blocking_repos.py --rosdistro kinetic --comp indigo --reposi turtlebot_interactions turtlebot_create_desktop kobuki_desktop
The following repos cannot be released because of unreleased dependencies:
	kobuki_desktop:
		driver_common
	turtlebot_interactions:
		kobuki_desktop
		smart_battery_msgs
		turtlebot_create_desktop
The following repos can be released, and are blocking other repos:
	turtlebot_create_desktop
```

The driver common dependency of kobuki_desktop is a little surprising and a quick browse on the codebase didn't find the dependency, but I might not have looked in all the right places.

Again smart_battery_msgs has been replaced so is not a blocker.

I released turtlebot_create_desktop but it needs a patch: https://github.com/turtlebot/turtlebot_create_desktop/issues/18

rohbotics, 2016-07-11:
For simulator:

```
./scripts/check_blocking_repos.py --rosdistro kinetic --comparison indigo --repositories turtlebot_simulator stdr_simulator
The following repos cannot be released because of unreleased dependencies:
	turtlebot_simulator:
		driver_common
		kobuki_desktop
		smart_battery_msgs
		stdr_simulator
The following repos can be released, and are blocking other repos:
	stdr_simulator
```

The driver_common dependency is weird here too, I can't seem to find it with grep.

kobuki_desktop will compile with this PR https://github.com/yujinrobot/kobuki_desktop/pull/46

I opened an issue with stdr_simulator for a kinetic release https://github.com/stdr-simulator-ros-pkg/stdr_simulator/issues/193

rohbotics, 2016-08-12:
Looks like we are good for turtlebot_interactions

```
./scripts/check_blocking_repos.py --rosdistro kinetic --comp indigo --reposi turtlebot_interactions turtlebot_create_desktop kobuki_desktop
Ignoring inputs which have already been released:
	kobuki_desktop
	turtlebot_create_desktop
The following repos cannot be released because of unreleased dependencies:
	turtlebot_interactions:
		smart_battery_msgs
```

For turtlebot_simulator:

```
./scripts/check_blocking_repos.py --rosdistro kinetic --comparison indigo --repositories turtlebot_simulator stdr_simulator
Ignoring inputs which have already been released:
	stdr_simulator
The following repos cannot be released because of unreleased dependencies:
	turtlebot_simulator:
		driver_common
		smart_battery_msgs
```

I am still not sure where driver_common is coming from

tfoote, 2016-08-12:
turtlebot_interactions released https://github.com/ros/rosdistro/pull/12350

tfoote, 2016-08-12:
Those packages have built at least on xenial the other builds are in progress. Thanks for everyone's effort to get this out!

closed 2016-08-12 by tfoote