turtlebot/turtlebot #221: Verify Astra URDF

bit-pirate, 2016-06-01:
The internal links need to be verified.

stonier, 2016-11-15:
I am guessing this means the Turtlebot [REP 119](http://www.ros.org/reps/rep-0119.html) related settings.

For reference, the astra urdf went in at pull request #220.

Two points here:

* Right now it does not use the properities in `turtlebot_properties.urdf` (e.g. `cam_px`), it should.
* Need to check the depth/cam relative links to make sure they are in the correct place according to the REP
* Finally, check that there is no field of view concerns with these

bit-pirate, 2016-11-15:
@stonier we should at Hyunchul to the TB organisation so we can assign issues like this one to him.