turtlebot/turtlebot #234: Velocity smoother for Teleop non compatible with namespaces

lc75, 2016-10-26:
It seems that the Velocity Smoother for teleop is not compatible with namespaces. If I use the smoother inside a group namespace \<group ns="robot1">, it will not work and use the wrong topics.

My suggestion is to avoid using absolute topic names in the \<remap> tags and for the nodelet manager.

`turtlebot_teleop/launch/includes/velocity_smoother.launch.xml` should be like this:

\
\<launch>
\<node pkg="nodelet" type="nodelet" name="teleop_velocity_smoother" args="load yocs_velocity_smoother/VelocitySmootherNodelet mobile_base_nodelet_manager">
\<rosparam file="$(find turtlebot_bringup)/param/defaults/smoother.yaml" command="load"/>
\<remap from="teleop_velocity_smoother/smooth_cmd_vel" to="cmd_vel_mux/input/teleop"/>

\
\<remap from="teleop_velocity_smoother/odometry" to="odom"/>
\<remap from="teleop_velocity_smoother/robot_cmd_vel" to="mobile_base/commands/velocity"/>
\</node>
\</launch>

Same issue is present in:
`turtlebot_follower/launch/includes/velocity_smoother.launch.xml`
`turtlebot_navigation/launch/includes/velocity_smoother.launch.xml`

stonier, 2016-10-26:
PR's are good :)

lc75, 2016-10-26:
I'm sorry Daniel, but I don't understand your comment. Is this an issue or should it be left as it is now?

stonier, 2016-10-26:
It's a reasonable improvement...send us a Pull Request

closed 2016-10-26 by lc75