turtlebot/turtlebot #236: Is there an easy way to start with Gazebo in Kinetic?

alesolano, 2016-11-04:
Hi!
I'm new in kinetic and Gazebo. I upgraded to kinetic when I was just starting to learn to use Gazebo, because I read that kinetic uses Gazebo 7.x versions, but I'm having problems to install the turtlebot packages.

I know that in indigo you could do:
`$ sudo apt-get install ros-indigo-turtlebot ros-indigo-turtlebot-apps ros-indigo-turtlebot-interactions ros-indigo-turtlebot-simulator ros-indigo-kobuki-ftdi ros-indigo-rocon-remocon ros-indigo-rocon-qt-library ros-indigo-ar-track-alvar-msgs`

And then just with this:
`$ roslaunch turtlebot_gazebo turtlebot_world.launch`
and
`$ roslaunch turtlebot_teleop keyboard_teleop.launch`
you could be moving the turtlebot easily.

I tried to install all the packages from above, changing _indigo_ for _kinetic_, but I couldn't install the rocon packages, and now I can't launch the turtlebot_world.launch in gazebo.
```
Invalid <arg> tag: environment variable 'TURTLEBOT_GAZEBO_WORLD_FILE' is not set. 

Arg xml is <arg default="$(env TURTLEBOT_GAZEBO_WORLD_FILE)" name="world_file"/>
The traceback for the exception was written to the log file
```

Should I go back to indigo?

Thanks!

stonier, 2016-11-10:
You should be able to get what you need just by installing `ros-kinetic-turtlebot-gazebo`, which will take care of all the dependencies.

Once you install them, make sure you rerun `source /opt/ros/kinetic/setup.bash` to catch the `TURTLEBOT_GAZEBO_WORLD_FILE` variable.

closed 2016-11-10 by stonier

ScitechBot, 2017-03-25:
I just followed this stream of conversation and like alesolano I am a noob.  I have a fresh install of kinetic and I used ros-kinetic-turtlebot-gazebo and source /opt/ros/kinetic/setup.bash to install turtle bot. I ran roslaunch turtlebot_gazebo turtlebot_world.launch and gazebo did launch, but without a predefined world.  What have I missed?

ScitechBot, 2017-03-26:
Executed another restart and gazebo popped quickly open and with the turtlebot world displayed.  So like all technologies, "have you turned it off and on again".

BrainPotter, 2018-05-03:
Hello,I want to ask why I ran "roslaunch turtlebot_teleop keyboard_teleop.launch"and I can see the program can run properly,but when I click "i",my robot in gazebo did not move?My OS is ubuntu16.04.kinetic and turtlebot.

tfoote, 2018-05-04:
@BrainPotter Please ask your question on answers.ros.org following our support guidelines: http://wiki.ros.org/Support

When you do so please make sure to include enough information to reproduce your problem in full.

Nathanaelgithub, 2018-07-27:
Running this

`export TURTLEBOT_GAZEBO_WORLD_FILE=/opt/ros/kinetic/share/turtlebot_gazebo/worlds/playground.world`

before launching the world worked for me.