turtlebot/turtlebot #250: cannot launch kobuki_dashboard on kinetic

fishcu, 2017-01-30:
Hi, I installed turtlebot kinetic packages using apt-get. I'm getting some errors, as I already described on ROS answers:
http://answers.ros.org/question/252837/cannot-launch-kobuki_dashboard-on-kinetic/?comment=253230#post-id-253230
http://answers.ros.org/question/252931/turtlebot-debs-installation-missing-kinetic-packages/

The main issue(s) seem to be:
- When trying "rqt -s kobuki_dashboard", I get "ImportError: cannot import name QMessageBox"
-  If I "fix" the file in question, "dashboard.py", by changing line 8 to:
"from PyQt4.QtGui import QMessageBox, QAction"
Then python will segfault.
- Besides that, I cannot find these 2 packages: "ros-kinetic-rocon-remocon" and "ros-kinetic-rocon-qt-library". Maybe it is related.

yuki-asano, 2017-02-21:
Hi, I'm facing the same problem when setting up a turtlebot using apt-get on ubuntu 16.04 and ros kinetic.
I hope this bug will be fixed.

yuki-asano, 2017-02-22:
This problem is related to qt5 used on ubuntu16.04.
Thanks for your advice. @k-okada

As correcting following two lines, it looks like working as shown in the attached screenshot.

in dashboard.py
from QtGui import QMessageBox, QAction 
-> 
from QtWidgets import QMessageBox, QAction

in battery_widgets.py
from python_qt_binding.QtGui import QToolButton
->
from QtWidgets import QToolButton

[image]

stonier, 2017-02-23:
Coming in 0.7.3, thanks @yuki-asano

closed 2017-02-23 by stonier