turtlebot/turtlebot #253: “turtlebot_teleop keyboard_teleop.launch”can't control turtlebot in gazebo

wangzhaodong123, 2017-05-06:
When I "roslaunch turtlebot_gazebo turtlebot_world.launch",then "roslaunch turtlebot_teleop keyboard_teleop.launch",I found that keyboard can't control the turtlebot.
Can you tell me why?Thank you.

tfoote, 2017-05-08:
This is a question that is more appropriate for http://answers.ros.org following our [Support Guidelines](http://wiki.ros.org/Support)

Please make sure to provide enough information in your question to reproduce your problem. To help you we need to be able to know exactly what you're doing otherwise we will only be guessing what's wrong.

closed 2017-05-08 by tfoote

wangzhaodong123, 2017-05-14:
After installing libgazebo7-dev, I have solved this problem. Thank you.