turtlebot/turtlebot #270: No such file or directory: .../kobuki_circles_kinect.urdf.xacro

luoxiaojie9527, 2018-07-11:
I launched a command: roslaunch turtlebot_bringup minimal.launch to launch turtlebot2 indigo. It occured an error:
Traceback (most recent call last):
File "/opt/ros/indigo/share/xacro/xacro.py", line 62, in <module>
xacro.main()
File "/opt/ros/indigo/lib/python2.7/dist-packages/xacro/__init__.py", line 673, in main
f = open(args[0])
IOError: [Errno 2] No such file or directory: '/home/turtlebot/turtlebot/src/turtlebot/turtlebot_description/robots/kobuki_circles_kinect.urdf.xacro'
while processing /home/turtlebot/turtlebot/src/turtlebot/turtlebot_bringup/launch/includes/robot.launch.xml:
while processing /home/turtlebot/turtlebot/src/turtlebot/turtlebot_bringup/launch/includes/description.launch.xml:
Invalid <param> tag: Cannot load command parameter [robot_description]: command [/opt/ros/indigo/share/xacro/xacro.py '/home/turtlebot/turtlebot/src/turtlebot/turtlebot_description/robots/kobuki_circles_kinect.urdf.xacro'] returned with code [1]. 

Param xml is <param command="$(arg urdf_file)" name="robot_description"/>
The traceback for the exception was written to the log file
I couldn't find the file: kobuki_circles_kinect.urdf.xacro. How to fix this problem?

VeryHardBit, 2021-11-17:
> I launched a command: roslaunch turtlebot_bringup minimal.launch to launch turtlebot2 indigo. It occured an error: _Traceback (most recent call last): File "/opt/ros/indigo/share/xacro/xacro.py", line 62, in xacro.main() File "/opt/ros/indigo/lib/python2.7/dist-packages/xacro/**init**.py", line 673, in main f = open(args[0]) IOError: [Errno 2] No such file or directory: '/home/turtlebot/turtlebot/src/turtlebot/turtlebot_description/robots/kobuki_circles_kinect.urdf.xacro' while processing /home/turtlebot/turtlebot/src/turtlebot/turtlebot_bringup/launch/includes/robot.launch.xml: while processing /home/turtlebot/turtlebot/src/turtlebot/turtlebot_bringup/launch/includes/description.launch.xml: Invalid tag: Cannot load command parameter [robot_description]: command [/opt/ros/indigo/share/xacro/xacro.py '/home/turtlebot/turtlebot/src/turtlebot/turtlebot_description/robots/kobuki_circles_kinect.urdf.xacro'] returned with code [1].
> 
> Param xml is The traceback for the exception was written to the log file_ I couldn't find the file: kobuki_circles_kinect.urdf.xacro. How to fix this problem?

I'm also facing the same error.How did you fix it?

luoxiaojie9527, 2021-12-10:
It has been a long time and I almost forgotten how I fixed it. Maybe you can set the environment variable of the turtlebot type. I remember there are two types of turtlebot2. One is Circle, the other is Hexagon.