turtlebot/turtlebot #272: Melodic release?

skruzic, 2018-10-24:
Is there Melodic release to be released soon? I tried Kinetic release with Melodic, and it does not work (I tried Gazebo simulation for Turtlebot).

130s, 2018-11-28:
I'm interested in contributing to this in my spare time, as I have a Ubuntu 18.04 laptop sitting on top of my Turtlebot**2** (no Turtlebot3 on my end :/ ).

There are at least a few discussions here and there regarding this.
- [answers.ros.org#q294600](https://answers.ros.org/question/294600/ros-melodic-does-it-support-turtlebot2)
- [discourse.ros.org#6069](https://discourse.ros.org/t/turtlebot-2-for-melodic/6069/3)

With reviewing these threads, I can see at least the following might need to be done (maintainers, if you agree please copy/modify these and add to OP so that tracking on github will be easier).

- Clear all missing depedency.
  - [ ] [depthimage_to_laserscan](http://wiki.ros.org/depthimage_to_laserscan) is still missing. Possible blocker https://github.com/ros-perception/depthimage_to_laserscan/issues/31
- Build turtlebot packages from source.
  - What set of packages is a minimum? Those defined in [this turtlebot.rosinstall](https://github.com/turtlebot/turtlebot/blob/3f8c66951a7d3146ec8c6b947747dc13f7a8aaee/turtlebot.rosinstall) file?
- Hardware compatibility check. What Turtlebot h/w needs compatible?
  - [ ] Turtlebot2
  - [ ] Turtlebot3 (I have no idea if Turtlebot3 utilizes the software in this repo).

lukewebdev, 2018-11-28:
I would like to help with this also.  Turtlebot2 on Melodic would be very helpful.

stonier, 2018-11-28:
@tfoote Might be worth transferring maintainership of the various packages to @130s so he can get busy with the blooming part of this.

@corot are you still using the turtlebots? Would you be able to release the `yujin_ocs`, `kobuki`, `kobuki_desktop` packages?

130s, 2018-12-02:
Found more missing dependenies:

- [ ] stdr_resources
- [ ] world_canvas_server
- [ ] rospy_message_converter
- [ ] std_capabilities
- [ ] gmapping
- [ ] robot_pose_ekf
- [ ] capabilities
- [ ] std_capabilities
- [ ] depthimage_to_laserscan

```
$ rosdep install -r -y --from-paths src --ignore-src
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:                                                       
turtlebot_stdr: Cannot locate rosdep definition for [stdr_resources]
turtlebot_rapps: Cannot locate rosdep definition for [world_canvas_server]
yocs_virtual_sensor: Cannot locate rosdep definition for [rospy_message_converter]
kobuki_capabilities: Cannot locate rosdep definition for [std_capabilities]
turtlebot_navigation: Cannot locate rosdep definition for [gmapping]
turtlebot_gazebo: Cannot locate rosdep definition for [robot_pose_ekf]
kobuki_node: Cannot locate rosdep definition for [capabilities]
turtlebot_capabilities: Cannot locate rosdep definition for [std_capabilities]
turtlebot_bringup: Cannot locate rosdep definition for [depthimage_to_laserscan]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully
```

130s, 2018-12-02:
Turned out `robot_pose_ekf` won't be available on Melodic ros-planning/navigation#701, where alternatives are suggested. Unless someone adds changes the binary release for Melodic won't happen I'm afraid.

spragunr, 2018-12-11:
It looks like `robot_pose_ekf` is only used for Create-based Turtlebots.  As long as the other dependencies can be dealt with, it would be straightforward to make a Kobuki-only release for Melodic.

lukewebdev, 2018-12-11:
Are there any other packages that are missing for Create-based Turtlebots? Maybe I can help port them?  The Create is still being marketed and sold so it might be nice to have.

tfoote, 2018-12-12:
@lukewebdev The original create support is for the Create 1 which is long since EOL and similar vintage to the robot_pose_ekf. I don't believe that the Create2 has been used in a core TurtleBot2 deployment. Are you using the main launch file for you Create2 based Turtlebot? ala https://github.com/turtlebot/turtlebot/blob/99bf4e6c6a8e17e0840ec373f0016c957480acee/turtlebot_bringup/launch/includes/create/mobile_base.launch.xml

The other solution is not to spend time porting the old localization package, but to move the example forward to one of the newer replacements.

lukewebdev, 2018-12-12:
Yes, so far based on the tutorials,  I have been using 
roslaunch turtlebot_bringup minimal.launch
roslaunch turtlebot_navigation gmapping_demo.launch 

etc, and one I customized to add a YDLidar.

I've set a variable in my .bashrc to use the "create" mobile base, and is an argument which ensures the inclusion of /launch/includes/create/mobile_base.launch.xml

Everything has worked with kinect xbox 3d sensor, from mapping to teleoperation as expected.

I am in favor of moving the example forward to a newer replacement and am willing to help with this.  It's my first effort in contributing back to ROS so I may need to be pointed in the right direction first.

lukewebdev, 2018-12-12:
So I'll work on getting this to work with robot_localization/graft, is right?

130s, 2018-12-12:
@tfoote I appreciate your guidance!

Your post https://github.com/turtlebot/turtlebot/issues/272#issuecomment-446424536 sounds like you suggested 2 solutions but I'm not sure what was the first one.

For the 2nd suggestion,

> The other solution is not to spend time porting the old localization package, but to move the example forward to one of the newer replacements.

As I describe below, looks like `robot_pose_ekf` IS depended, not just used in tutorial.

It is only depended by 1) gazebo pkg, 2) create/roomba launch files. See below the search result I did against all source downloaded by [turtlebot.rosinstall](https://raw.githubusercontent.com/turtlebot/turtlebot/3f8c66951a7d3146ec8c6b947747dc13f7a8aaee/turtlebot.rosinstall).

To me the easiest is to create a melodic-onward branch, then remove the portion that depends on ekf pkg. Indeed it causes another branching, but assuming there will be less new develpment and the previous versions can be treated as legacy, so creating a new branch hopefully won't add much maintenance.

```
$ docker pull d130s/ros:melodic_turtlebot && docker run -it d130s/ros:melodic_turtlebot

# WSDIR=~/cws_tbot && mkdir -p $WSDIR/src && cd $WSDIR/src
# wget https://raw.githubusercontent.com/turtlebot/turtlebot/3f8c66951a7d3146ec8c6b947747dc13f7a8aaee/turtlebot.rosinstall && mv turtlebot.rosinstall .rosinstall
# cd $WSDIR && wstool update -t src
# ack -i robot_pose_ekf
turtlebot_simulator/turtlebot_gazebo/package.xml
20:  <run_depend>robot_pose_ekf</run_depend>

turtlebot_simulator/turtlebot_gazebo/launch/includes/create.launch.xml
14:  <node pkg="robot_pose_ekf" type="robot_pose_ekf" name="robot_pose_ekf">
16:    <remap from="robot_pose_ekf/odom" to="odom_combined"/>

turtlebot_simulator/turtlebot_gazebo/launch/includes/roomba.launch.xml
14:  <node pkg="robot_pose_ekf" type="robot_pose_ekf" name="robot_pose_ekf">
16:    <remap from="robot_pose_ekf/odom" to="odom_combined"/>

turtlebot_simulator/turtlebot_gazebo/CHANGELOG.rst
36:* Do not use robot_pose_ekf for kobuki base.

kobuki_core/kobuki_driver/CHANGELOG.rst
77:* Do not use robot_pose_ekf; use imu for heading and encoders for position.

kobuki/kobuki_node/param/base.yaml
23:# Causes node to publish TF for odom_frame to base_frame. Disable only if you plan to use robot_pose_ekf
29:# way, for example filtering and fussing with robot_pose_ekf (bool, default: true)

kobuki/kobuki_node/src/library/odometry.cpp
57:      ROS_INFO_STREAM("Kobuki : not publishing transforms (see robot_pose_ekf) [" << name << "].");
67:      ROS_INFO_STREAM("Kobuki : using encoders for heading (see robot_pose_ekf) [" << name << "].");
145:  // Pose covariance (required by robot_pose_ekf) TODO: publish realistic values
154:  odom->pose.covariance[28] = DBL_MAX; // is a requirement of robot_pose_ekf

kobuki/kobuki_node/src/library/slot_callbacks.cpp
131:      // set a non-zero covariance on unused dimensions (pitch and roll); this is a requirement of robot_pose_ekf
141:      // angular velocity covariance; useless by now, but robot_pose_ekf's

kobuki/kobuki_node/CHANGELOG.rst
69:* Do not use robot_pose_ekf. Instead, use imu for heading and encoders por x and y. However, parameter use_imu_heading makes trivial to switch back to the previous system.
73:* Do not use robot_pose_ekf; use imu for heading and encoders for position.

kobuki/kobuki_description/CHANGELOG.rst
50:* Do not use robot_pose_ekf. Instead, use imu for heading and encoders por x and y. However, parameter use_imu_heading makes trivial to switch back to the previous system.

turtlebot/turtlebot_bringup/package.xml
28:  <run_depend>robot_pose_ekf</run_depend>

turtlebot/turtlebot_bringup/launch/includes/roomba/mobile_base.launch.xml
25:  <node pkg="robot_pose_ekf" type="robot_pose_ekf" name="robot_pose_ekf">

turtlebot/turtlebot_bringup/launch/includes/create/mobile_base.launch.xml
28:  <node pkg="robot_pose_ekf" type="robot_pose_ekf" name="robot_pose_ekf">
30:    <remap from="robot_pose_ekf/odom" to="odom_combined"/>

turtlebot/turtlebot_bringup/CHANGELOG.rst
207:* Do not use robot_pose_ekf for kobuki base. Instead, use imu for heading and encoders por x and y.
```

mintar, 2019-01-17:
rospy_message_converter is released into melodic and should be available in the next sync: uos/rospy_message_converter#21

mintar, 2019-01-17:
BTW, if you don't know it, here's a nice web page that helps to track blocking dependencies:

http://repositories.ros.org/status_page/blocked_releases_lunar.html?q=id%3D%22turtlebot%22&s=6&r=1

Unfortunately, it only lists packages which have been released into the previous distro. So turtlebot is only listed for lunar (not melodic), since kinetic is the latest distro into which turtlebot has been released.

bunchofcoders, 2019-03-09:
I was having issues on ubuntu 18.04 with melodic. I got it to work after building packages one by one. Here is what I did and worked for me. Hope this helps.

turtlebot_simulator
turtlebot
turtlebot_apps
robot_pose_ekf
depthimage_to_laserscan
kobuki_msgs
kobuki_desktop
orocos-bayesian-filtering
turtlebot_msgs
joystick_drivers
kobuki_description
kobuki_bumper2pc
yocs_cmd_vel_mux
turtlebot_interactions

turtlebot_simulator
 git clone https://github.com/turtlebot/turtlebot_simulator

turtlebot
 git clone https://github.com/turtlebot/turtlebot.git

turtlebot_apps
 git clone https://github.com/turtlebot/turtlebot_apps.git

robot_pose_ekf
 git clone https://github.com/udacity/robot_pose_ekf 

depthimage_to_laserscan
 git clone https://github.com/ros-perception/depthimage_to_laserscan.git

kobuki_msgs
 git clone https://github.com/yujinrobot/kobuki_msgs.git

kobuki_desktop
 git clone https://github.com/yujinrobot/kobuki_desktop.git
 cd kobuki_desktop/
 rm -r kobuki_qtestsuite

install opencv follow instructions given below since bayesian filter depends on it
 https://docs.microsoft.com/en-us/cognitive-toolkit/setup-opencv-on-linux
 if you get stdlib error use -DENABLE_PRECOMPILED_HEADERS=OFF flag on cmake

orocos-bayesian-filtering
 git clone https://github.com/toeklk/orocos-bayesian-filtering.git

cd ..
catkin_make

if you get errors here try building them individually
 cd src/orocos-bayesian-filtering/
 make
 make install

 cd orocos_bfl/
 ./configure 
 make
 make install
 sudo make install

turtlebot_msgs
 cd ~/source/workspaces/robotics/catkin_ws/src
 git clone https://github.com/turtlebot/turtlebot_msgs.git

joystick_drivers
 first install following dependencies for joystick drivers

 sudo apt-get install libusb-dev

 sudo apt-get install libspnav-dev

 sudo apt-get install ros-melodic-joystick-drivers

 sudo apt-get install bluetooth

 sudo apt-get install libbluetooth-dev

 sudo apt-get install libcwiid-dev

 git clone https://github.com/ros-drivers/joystick_drivers.git

kobuki_description

 clone entire kobuki repository some where 

 cd ~/repos/
 git clone https://github.com/yujinrobot/kobuki.git

 copy only  description package to your workspace src directory
 cd kobuki
 cp -r kobuki_description ~/source/workspaces/robotics/catkin_ws/src/

At this point catkin_make error should be gone and you should be able to roslaunch turtlebot_gazebo but with lots of red error on console. These are due to some missing nodes in following packages.

kobuki_bumper2pc
 copy from kobuki repo

 cd ~/source/workspaces/robotics/catkin_ws/src/
 cp -r ~/repos/kobuki/kobuki_bumper2pc .

turtlebot_interactions
 git clone https://github.com/turtlebot/turtlebot_interactions.git 

yocs_cmd_vel_mux
 clone entire repo somewhere 
 cd ~/repos
 git clone https://github.com/yujinrobot/yujin_ocs.git

 copy only cmd_vel_mux

 cd ~/source/workspaces/robotics/catkin_ws/src/
 cp -r ~/repos/yujin_ocs/yocs_cmd_vel_mux/ .

 cd ..
 catkin_make

ProfJust, 2019-03-12:
We have it all installed on a clean Ubuntu 18.04 and ROS Melodic NUC-PC, but we cannot start our turtlebot. We always get this Error-Message.

~$ roslaunch turtlebot_bringup minimal.launch
... logging to /home/turtle/.ros/log/75c3c312-44bc-11e9-a937-0016eb38839e/roslaunch-turtleBot-170405A-14617.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
Resource not found: kobuki_node
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/turtle/catkin_ws/src
ROS path [2]=/opt/ros/melodic/share
The traceback for the exception was written to the log file

Can you please tell me how we can get the missing resources?

bunchofcoders, 2019-03-12:
Please refer my post above for step by step instruction to install and build all the dependencies.
For kobuki_msgs, in your /home/turtle/catkin_ws/src directory clone repository and run catkin_make, 
cd /home/turtle/catkin_ws/src
git clone https://github.com/yujinrobot/kobuki_msgs.git
cd ..
catkin_make

Hope this helps.

corot, 2019-03-14:
> @corot are you still using the turtlebots? Would you be able to release the `yujin_ocs`, `kobuki`, `kobuki_desktop` packages? I'm afraid I'm short on time and out of the country for a while.
Hi @stonier, I'm trying to release `kobuki`, `kobuki_desktop` packages, but they depend on `rocon_apps` and `rocon_app_manager` :crying_cat_face: Could you scratch some time to release ROCON?
btw, `yujin_ocs` is already released/

ProfJust, 2019-03-19:
Ok, now it works, yeahh!

Do everything that bunchofcoders posts above, then add ... 
1.) Install all Kobuki-packages
    $ sudo apt-get install ros-melodic-kobuki-*

2.) Clone ecl_stream to catkin:_ws
   $ cd ~catkin_ws/src
   $ sudo apt-get install ros-melodic-ecl-streams

3.) Copy the folders  kobuki_node, kobuki_keyop, kobuki_safety_controller, yocs_controllers
from the cloned directory to catkin_ws/src
    $  cd ~catkin_ws
    $ catkin_make

lukewebdev, 2019-03-22:
I can confirm the combination of ProfJust and bunchofcoders steps allowed me to successfully run catkin_make.

Some things I did differently:
1. After cloning the repos I ran: ls | xargs -I{} git -C {} pull in order to recursively pull some because I had checked some out a long time ago.
2. This helped me notice some had new branches, some melodic, some kinetic. Many were still on indigo. I switched to the latest branch available.
3. I had already cloned turtlebot_create and turtlebot_create_desktop into /src and this does not seem to had any negative effect.

I have both a Kobuki Turtlebot 2 and  a Create2 (irobot) Turtlebot.

I ran these steps on a rockpro64  which I plan to use on the Create 2 to replace its raspberry pi model 3 b which is very slow.

I will try this out on the actual turtlebot(s) ASAP and update if any issues running the launch files.

gaunthan, 2019-03-22:
Thanks to the work of bunchofcoders and ProfJust! 

I have wrapped there things up for convenience, and tested on Ubuntu 18.04 with ROS melodic.

Check it out, [gaunthan/Turtlebot2-On-Melodic](https://github.com/gaunthan/Turtlebot2-On-Melodic)

Some packages listed above are not necessary, so I do not include them at the building script.

There are two scripts inside the repository

- `build_basic.sh` only install necessary packages that allow you to use keyboard to control Turtlebot2
- `build_all.sh` install everything that listed by bunchofcoders and ProfJust

Choose one you prefer to.

lukewebdev, 2019-03-22:
Looks good, I will try this out. So will this somehow get rolled into Melodic so it works by default or what is the future of that?

gaunthan, 2019-03-23:
> Looks good, I will try this out. So will this somehow get rolled into Melodic so it works by default or what is the future of that?

See https://answers.ros.org/question/294600/ros-melodic-does-it-support-turtlebot2/?answer=294603#post-id-294603

lukewebdev, 2019-03-25:
@tfoote Just to follow up on the using the Create 2 irobot in Kinetic, here are the only steps I had to take in that version to make it work:

http://wiki.ros.org/turtlebot_calibration/Tutorials/Calibrate%20Odometry%20and%20Gyro
http://wiki.ros.org/turtlebot/Tutorials/indigo/Create%20Base

So it was very easy to do.

I'd been using it for about 9 months when I decided to try to upgrade to melodic so it would work with a new board that only runs 18.04.

corot, 2019-03-26:
Nice work! Thanks

From my part, I have created a melodic branch with the minimum changes required to make things work. @gaunthan, could you check it out in your scripts to verify all works?

@tfoote, as we are all very busy here, I have opted for directly pushing changes instead of PRequesting one by one. Hopefully is ok for you; but otherwise I can remove the branch and start anew with PRs

lukewebdev, 2019-03-26:
I will try out this branch now on both my create2 and kobuki Turtlebot2.

lukewebdev, 2019-03-26:
@corot On a fresh source install of Melodic that I did last night, I checked out this turtlebot melodic branch and got this error when performing catkin_make:

CMake Error at /home/rock64/ros_catkin_ws/install_isolated/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "joy" with any of
  the following names:

    joyConfig.cmake
    joy-config.cmake

lukewebdev, 2019-03-27:
Good news, I was able to get this to launch properly with the Create 2.

The most notable thing missing was robot_pose_ekf but we knew that was the case already and I addressed it with the steps below.

These are the steps I took:
(cd into the catkin workspace)
sudo apt-get install ros-melodic-joy*
sudo apt-get install ros-melodic-urdf*
sudo apt-get install ros-melodic-*kobuki*
cd src
git clone https://github.com/lukewebdev/turtlebot_create.git
cd ../
sudo apt-get install ros-melodic-diagnostics
sudo apt-get install ros-melodic-nodelet
sudo mkdir ~/repos
cd ~/repos
git clone -b kinetic-devel https://github.com/ros-planning/navigation.git
cd ~/turtlebot_melodic_ws
ln ~/repos/navigation/robot_pose_ekf robot_pose_ekf -s
sudo apt-get install ros-melodic-bfl
sudo apt install python3-serial
sudo apt-get install python-serial
Sudo apt-get install ros-melodic-yujin-ocs
sudo apt-get install ros-melodic-python-orocos-kdl
roslaunch turtlebot_bringup minimal.launch

At this point, there were no errors other than laptop battery (which in my experience is normal for Create 2) and ....I can't connect to the create 2 sci at all.  

However, I had this problem regularly with the kinetic install I used previously

To ensure it was not my usb cable or my battery, I was able to run an unrelated python library http://www.irobot.com/~/media/MainSite/Files/About/STEM/Create/Create2_TetheredDrive.py and it connected and controlled the Create 2 base every time consistently and without delay. So that tells me something is still amiss with the connection part, but nothing that was not already problematic in kinetic.

The exact error is [WARN] [1553656093.298154]: Create : robot not connected yet, sci not available

I will run this on the Kobuki base next! Thanks!

lukewebdev, 2019-03-27:
I can confirm that the problem that has  been inexplicably plaguing my Create2  usage in Kinetic was the create_driver.py file, it had baud rates that (sometimes/mostly/but-not-always) prevent it from connecting!  I set it to only use 115200 and it connected right away.  I will refactor this driver to account for Create 1 Vs Create 2 baud rates and do a PR asap.  I was actually unaware that this driver was never supposed to work with Create 2 because I used it many times although often had trouble connecting unless I disconnected the battery and quickly connected.  With this change, I was able to teleop etc but have not tested the other demos. https://github.com/lukewebdev/turtlebot_create/tree/melodic-create-2

Kobuki testing tomorrow.

lukewebdev, 2019-03-27:
PRs to should fix the above Create vs Create 2 vs Roomba baudrate issues with least amount of changes:
https://github.com/turtlebot/turtlebot/pull/275
https://github.com/turtlebot/turtlebot_create/pull/41

cdalizadeh, 2020-01-19:
For anyone following bunchofcoders' guide above, changes have been made to https://github.com/yujinrobot/kobuki_msgs.git that prevent catkin_make from building. I managed to get it working by checking out commit 768d4d0 in that repository, from before the changes.

stonier, 2020-01-19:
Kobuki msgs does have a release/0.7-melodic branch. If you work off that you will get future fixes unlike fixing yourself to a branch

cdalizadeh, 2020-01-19:
Ah, thanks for the tip. I'll switch to that.

lukewebdev, 2020-02-18:
I was able to get map creation and navigation running on kobuki turtlebot 2 on 18.04/melodic (with a bunch of kinetic code in src folder) loaded on a Jetson Nano, but with a PC running 16.04/kinetic.  It was a convoluted effort though and I'm not sure exactly how to reproduce.

I am interested in trying ROCON next.

alexantosh, 2020-08-20:
Alright, clearly, the people here have tried a bunch of things. Is there a single post or blog that explains how I can have turtlebot2 based on kobuki running on ros-melodic? 
For now even gazebo support of the turtlebot2 can suffice? 
Thanks.

civerachb-cpr, 2020-08-24:
Two years later: are there any plans to bloom the Turtlebot packages for Melodic?  Kinetic is going EOL in the near-future and it would be nice to not have to build from source.

hanzheteng, 2020-09-09:
Same. We hope to see official release on Melodic, as we really need them for robotics instruction at school.

QianleLi, 2020-09-10:
Need official release +1

bunchofcoders, 2020-09-12:
I was using turtlebot3 on melodic last year, had to build everything myself as melodic was new. Hope this helps, while waiting..

https://github.com/bunchofcoders/turtlebot3-melodic

lukewebdev, 2020-09-12:
I think this thread is about turtlebot 2.

civerachb-cpr, 2020-09-12:
> I was using turtlebot3 on melodic last year, had to build everything myself as melodic was new.

This thread is about the Melodic release of Turtlebot2. Yes, it's possible to be build from source, but the request here is to officially release pre-built packages that can be installed on e.g. Ubuntu 18.04 via apt-get install.

Melodic is now 2 years old and Kinetic is going EOL early next year. But the latest official packages are for Kinetic targeting Ubuntu 16.04.

stonier, 2020-09-12:
Seems there is enough interest - I'd suggest perhaps forming a working group to do the releasing yourselves.

alexantosh, 2020-09-12:
@stonier am interested. Especially if you are ready for being consulted on very random topics and issues :). Since am pretty new to cmake management platform.

hanzheteng, 2021-04-09:
Kinetic is approach EOL by the end of this month. Do we have any updates on the official Melodic release? I am looking forward to it with many thanks.