turtlebot/turtlebot #277: Why does the turtlebot2 have a small angular velocity acceleration?

LingyunLL, 2019-06-26:
When we used the turtlebot2 of the kobuki chassis, we found that when the control speed is sent to the /mobile_base/command/velocity topic, the angular velocity increases particularly slowly, which is much slower than the linear velocity increase. It takes about several seconds. How can I solve this problem?

tfoote, 2019-07-10:
The base passes commands through a low pass filter to avoid jerky motions that can cause jerky motions that cause the robot to wobble. You can tune the parameters to be more aggressive if you'd like. The angular and linear velocities are different parameters in different units they need to be tuned separately.

closed 2019-07-10 by tfoote

LingyunLL, 2019-07-15:
Thank you for your answer, but I still can't find where to tune the angular acceleration. Do you mean different units like gyroscopes, accelerometers, etc? I found that velocity_smoother contains the accel_lim_w parameter associated with the angular acceleration, but it has no effect after the adjustment. In addition, I found the kobuki_drive package, which may have related parameters, but I did not use this package.could you continue to find out, thank you for your patience.