turtlebot/turtlebot #281: Odom value error

thomastom11, 2020-04-22:
I'm using turtlebot 2 with kobuki base. I have launched a world file with turtlebot, the turtlebot came in its default positiom (0,0,0). Now i want my robot to be at (-1.6,0,0). I have made the changes accordingly and now the robot is in required position, but when i checked the odom value it is showing that position as (0,0,0). why it is showing like that?? Do i need to make some other change?? I want that odom value to be same as that of initial position that i have given.

mikeferguson, 2020-04-22:
That's generally not how odometry is setup in ROS - the odom position is where the robot started up, when you localize the robot the offset goes into the odom->map transform (usually broadcast by AMCL).

thomastom11, 2020-04-22:
But the same case was working with turtlebot3 why it is not working in turtlebot 2? In turtlebot3 simulation, the initial pose of robot was (0,0,0), i made it to (-1.6,0,0) and then i checked the odom value it was showing the same value (-1.6,0,0) this case is not working for turtlebot 2.

mikeferguson, 2020-04-22:
How did you "made it to (-1.6, 0, 0)" - if you did it using the "set 2d pose" in RVIZ, then I'd say that sounds like some bug in turtlebot3 (because that should affect the map->odom, not odom->base_link).

Also, it's worth noting, the turtlebot2 and turtlebot3 have very little in common other than their name and use of standard higher-level applications (e.g. navigation stack).

thomastom11, 2020-04-22:
Actually i was speaking about the simulation part which is done in gazebo where navigation stack is not used. Initially when i launched gazebo the turtlebot2 and 3 was in 0,0,0 pose. For t3 i made the changes in launch file where x_pos,y_pos,z_pos where three arguments and these three are passed to spawn model but in t2 case ROBOT_INITIAL_POSE is an env file so i exported it like this:- export ROBOT_INITIAL_POSE="-x -1.6 -y 0 -z 0". After doing this i launched the simulation file, robot moved to my desired position but the odom value remains as 0,0,0

tfoote, 2020-05-21:
The odom value is initialized to zero at whatever location you start the odometry. As such it should be (0,0,0) whereever you spawn the robot. The TurtleBot 3 behavior is unexpected.