turtlebot/turtlebot #283: wheel_left_joint and wheel_right_joint are not in tf tree

willzoe, 2020-05-21:
When I use command `roslaunch turtlebot_bringup minimal.launch` to launch turtlebot, the terminal prints the following information:
```
... logging to /home/scuimg/.ros/log/6dadad54-9b92-11ea-98f7-4a0c02312ac9/roslaunch-scuimg-nano-18492.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

xacro: in-order processing became default in ROS Melodic. You can drop the option.
Deprecated: xacro tag 'kobuki' w/o 'xacro:' xml namespace prefix (will be forbidden in Noetic)
when processing file: /home/scuimg/turtlebot_ws/src/turtlebot/turtlebot_description/robots/kobuki_hexagons_asus_xtion_pro.urdf.xacro
Use the following command to fix incorrect tag usage:
find . -iname "*.xacro" | xargs sed -i 's#<\([/]\?\)\(if\|unless\|include\|arg\|property\|macro\|insert_block\)#<\1xacro:\2#g'

started roslaunch server http://192.168.0.115:46097/

SUMMARY
========

PARAMETERS
 * /bumper2pointcloud/pointcloud_radius: 0.24
 * /cmd_vel_mux/yaml_cfg_file: /home/scuimg/turt...
 * /diagnostic_aggregator/analyzers/input_ports/contains: ['Digital Input',...
 * /diagnostic_aggregator/analyzers/input_ports/path: Input Ports
 * /diagnostic_aggregator/analyzers/input_ports/remove_prefix: mobile_base_nodel...
 * /diagnostic_aggregator/analyzers/input_ports/timeout: 5.0
 * /diagnostic_aggregator/analyzers/input_ports/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/kobuki/contains: ['Watchdog', 'Mot...
 * /diagnostic_aggregator/analyzers/kobuki/path: Kobuki
 * /diagnostic_aggregator/analyzers/kobuki/remove_prefix: mobile_base_nodel...
 * /diagnostic_aggregator/analyzers/kobuki/timeout: 5.0
 * /diagnostic_aggregator/analyzers/kobuki/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/power/contains: ['Battery', 'Lapt...
 * /diagnostic_aggregator/analyzers/power/path: Power System
 * /diagnostic_aggregator/analyzers/power/remove_prefix: mobile_base_nodel...
 * /diagnostic_aggregator/analyzers/power/timeout: 5.0
 * /diagnostic_aggregator/analyzers/power/type: diagnostic_aggreg...
 * /diagnostic_aggregator/analyzers/sensors/contains: ['Cliff Sensor', ...
 * /diagnostic_aggregator/analyzers/sensors/path: Sensors
 * /diagnostic_aggregator/analyzers/sensors/remove_prefix: mobile_base_nodel...
 * /diagnostic_aggregator/analyzers/sensors/timeout: 5.0
 * /diagnostic_aggregator/analyzers/sensors/type: diagnostic_aggreg...
 * /diagnostic_aggregator/base_path: 
 * /diagnostic_aggregator/pub_rate: 1.0
 * /mobile_base/base_frame: base_footprint
 * /mobile_base/battery_capacity: 16.5
 * /mobile_base/battery_dangerous: 13.2
 * /mobile_base/battery_low: 14.0
 * /mobile_base/cmd_vel_timeout: 0.6
 * /mobile_base/device_port: /dev/kobuki
 * /mobile_base/odom_frame: odom
 * /mobile_base/publish_tf: True
 * /mobile_base/use_imu_heading: True
 * /mobile_base/wheel_left_joint_name: wheel_left_joint
 * /mobile_base/wheel_right_joint_name: wheel_right_joint
 * /robot/name: turtlebot
 * /robot/type: turtlebot
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 5.0
 * /rosdistro: melodic
 * /rosversion: 1.14.5
 * /turtlebot_laptop_battery/acpi_path: /sys/class/power_...
 * /use_sim_time: False

NODES
  /
    bumper2pointcloud (nodelet/nodelet)
    cmd_vel_mux (nodelet/nodelet)
    diagnostic_aggregator (diagnostic_aggregator/aggregator_node)
    mobile_base (nodelet/nodelet)
    mobile_base_nodelet_manager (nodelet/nodelet)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    turtlebot_laptop_battery (laptop_battery_monitor/laptop_battery.py)

auto-starting new master
process[master]: started with pid [18505]
ROS_MASTER_URI=http://192.168.0.115:11311

setting /run_id to 6dadad54-9b92-11ea-98f7-4a0c02312ac9
process[rosout-1]: started with pid [18516]
started core service [/rosout]
process[robot_state_publisher-2]: started with pid [18523]
process[diagnostic_aggregator-3]: started with pid [18524]
process[mobile_base_nodelet_manager-4]: started with pid [18525]
process[mobile_base-5]: started with pid [18531]
process[bumper2pointcloud-6]: started with pid [18536]
process[cmd_vel_mux-7]: started with pid [18539]
ERROR: cannot launch node of type [laptop_battery_monitor/laptop_battery.py]: laptop_battery_monitor
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/scuimg/myrobot/src
ROS path [2]=/home/scuimg/turtlebot_ws/src
ROS path [3]=/home/scuimg/catkin_ws/src
ROS path [4]=/opt/ros/melodic/share
[ WARN] [1590086387.478923809]: Kobuki : Robot firmware is outdated; we suggest you to upgrade it to benefit from the latest features. Consult how-to on: http://kobuki.yujinrobot.com/home-en/documentation/howtos/upgrading-firmware
[ WARN] [1590086387.479120531]: Kobuki : Robot firmware version is 1.1.4; latest version is 1.2.x
```

When I checked the tf tree, I did not find wheel_left_joint and wheel_right_joint in it.
[image]

Is this the same for yours?

corot, 2020-06-05:
Nop...  Can you echo /joint_states and filter `wheel_left_joint`?
e.g.:
`rostopic echo /joint_states | grep wheel_left_joint`

willzoe, 2020-06-05:
@corot When I echo /joint_states, nothing is displayed on the screen.

I think it may be because of some problems with the turtlebot related packages on my Jetson Nano. On my notebook, wheel_left_joint and wheel_right_joint can be displayed normally.

Now I want to install all the packages of turtlebot2 on Jetson Nano. The ROS version is melodic. Are there some good tutorials or blogs?

corot, 2020-06-05:
Yup, if nothing comes on joint states you are not getting robot status; I suppose you cannot get /odom nor any other topic coming from the robot.

For installing turtlebot on melodic, on issue #272 people discussed at length how to do it. For using a Jetson, I have no idea.

I will close this, as the problem seems unrelated to turtlebot.

closed 2020-06-05 by corot

lukewebdev, 2020-06-05:
> @corot When I echo /joint_states, nothing is displayed on the screen.
>
> I think it may be because of some problems with the turtlebot related packages on my Jetson Nano. On my notebook, [...]

I got this working on my Jetson nano a few months ago back in February. There were quite a few steps I had to take.  I have some notes in a google doc that might help.

willzoe, 2020-06-05:
@lukewebdev May I refer to your notes?

lukewebdev, 2020-06-06:
https://docs.google.com/document/d/1NpsWRT1ssieHXmME-v-8CC_8IEWo8qT9ccPDNBUbDwI/edit?usp=drivesdk

Any questions let me know.  The notes are a little rough but have everything I used to get it to work.

willzoe, 2020-06-06:
@lukewebdev When I use the command `curl -Lf https://raw.githubusercontent.com/lukewebdev/Turtlebot2-On-Melodic/lukewebdev-patch-1/install_all.sh | bash
` to perform the fifth step in the document, the terminal outputs the following information:
```
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  1144  100  1144    0     0    875      0  0:00:01  0:00:01 --:--:--   874
Cloning into 'turtlebot_simulator'...
remote: Enumerating objects: 13, done.
remote: Counting objects: 100% (13/13), done.
remote: Compressing objects: 100% (10/10), done.
remote: Total 1114 (delta 4), reused 9 (delta 3), pack-reused 1101
Receiving objects: 100% (1114/1114), 526.01 KiB | 46.00 KiB/s, done.
Resolving deltas: 100% (606/606), done.
Cloning into 'turtlebot'...
remote: Enumerating objects: 57, done.
remote: Counting objects: 100% (57/57), done.
remote: Compressing objects: 100% (33/33), done.
remote: Total 7656 (delta 37), reused 36 (delta 23), pack-reused 7599
Receiving objects: 100% (7656/7656), 10.05 MiB | 106.00 KiB/s, done.
Resolving deltas: 100% (4435/4435), done.
Cloning into 'turtlebot_apps'...
remote: Enumerating objects: 5811, done.
remote: Total 5811 (delta 0), reused 0 (delta 0), pack-reused 5811
Receiving objects: 100% (5811/5811), 2.99 MiB | 88.00 KiB/s, done.
Resolving deltas: 100% (3209/3209), done.
Cloning into 'turtlebot_msgs'...
remote: Enumerating objects: 47, done.
remote: Total 47 (delta 0), reused 0 (delta 0), pack-reused 47
Unpacking objects: 100% (47/47), done.
Cloning into 'turtlebot_interactions'...
remote: Enumerating objects: 862, done.
remote: Total 862 (delta 0), reused 0 (delta 0), pack-reused 862
Receiving objects: 100% (862/862), 169.72 KiB | 278.00 KiB/s, done.
Resolving deltas: 100% (504/504), done.
Cloning into 'kobuki_msgs'...
remote: Enumerating objects: 19, done.
remote: Counting objects: 100% (19/19), done.
remote: Compressing objects: 100% (16/16), done.
remote: Total 262 (delta 5), reused 10 (delta 3), pack-reused 243
Receiving objects: 100% (262/262), 35.49 KiB | 68.00 KiB/s, done.
Resolving deltas: 100% (139/139), done.
Already on 'release/0.7-melodic'
Your branch is up to date with 'origin/release/0.7-melodic'.
Cloning into 'kobuki_desktop'...
remote: Enumerating objects: 1900, done.
remote: Total 1900 (delta 0), reused 0 (delta 0), pack-reused 1900
Receiving objects: 100% (1900/1900), 432.95 KiB | 320.00 KiB/s, done.
Resolving deltas: 100% (1091/1091), done.
error: pathspec 'release/0.7-melodic' did not match any file(s) known to git.
Cloning into 'kobuki'...
remote: Enumerating objects: 28, done.
remote: Counting objects: 100% (28/28), done.
remote: Compressing objects: 100% (23/23), done.
remote: Total 19921 (delta 9), reused 8 (delta 3), pack-reused 19893
Receiving objects: 100% (19921/19921), 31.24 MiB | 92.00 KiB/s, done.
Resolving deltas: 100% (13699/13699), done.
bash: line 24: release/0.7-melodic: No such file or directory
Cloning into 'yujin_ocs'...
remote: Enumerating objects: 5156, done.
remote: Total 5156 (delta 0), reused 0 (delta 0), pack-reused 5156
Receiving objects: 100% (5156/5156), 3.94 MiB | 328.00 KiB/s, done.
Resolving deltas: 100% (3196/3196), done.
Branch 'release/0.8-melodic' set up to track remote branch 'release/0.8-melodic' from 'origin'.
Switched to a new branch 'release/0.8-melodic'
Cloning into 'orocos-bayesian-filtering'...
remote: Enumerating objects: 2390, done.
remote: Total 2390 (delta 0), reused 0 (delta 0), pack-reused 2390
Receiving objects: 100% (2390/2390), 787.06 KiB | 111.00 KiB/s, done.
Resolving deltas: 100% (1820/1820), done.
Orocos BFL configure script.
[... 718 lines cut ...]
-- ==> add_subdirectory(turtlebot_simulator/turtlebot_stdr)
-- +++ processing catkin package: 'kobuki_keyop'
-- ==> add_subdirectory(kobuki/kobuki_keyop)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- +++ processing catkin package: 'kobuki_auto_docking'
-- ==> add_subdirectory(kobuki/kobuki_auto_docking)
-- Using these message generators: gencpp;geneus;genlisp;gennodejs;genpy
-- +++ processing catkin package: 'depthimage_to_laserscan'
-- ==> add_subdirectory(depthimage_to_laserscan)
CMake Error at /opt/ros/melodic/share/image_geometry/cmake/image_geometryConfig.cmake:113 (message):
  Project 'image_geometry' specifies '/usr/include/opencv' as an include dir,
  which is not found.  It does neither exist as an absolute directory nor in
  '${{prefix}}//usr/include/opencv'.  Check the website
  'http://www.ros.org/wiki/image_geometry' for information and consider
  reporting the problem.
Call Stack (most recent call first):
  /opt/ros/melodic/share/catkin/cmake/catkinConfig.cmake:76 (find_package)
  depthimage_to_laserscan/CMakeLists.txt:6 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/nano/turtlebot_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/nano/turtlebot_ws/build/CMakeFiles/CMakeError.log".
Invoking "cmake" failed
```
Seems to have encountered some errors. Do you know how to resolve this error?

lukewebdev, 2020-06-07:
Looks like you are missing opencv or it is in the wrong place. Can you try installing it or find where it is installed? If it is already installed maybr make a symlink.

willzoe, 2020-06-07:
@lukewebdev What version of OpenCV are you using? 

The OpenCV version on the SD card is 3.3.1, but some blogs say that OpenCV3.2 is required to compile `depthimage_to_laserscan`.

When I tried to compile the CUDA version of OpenCV3.2 from the source code, the following compilation error occurred:
```
error: identifier "__shfl" is undefined
```
It seems to be because of the CUDA version, whose current version is 10.2.